Taro-Hayashi/KillerWhale #3: 側面ユニット支柱の形状について

nerinanarine, 2023-08-12:
部品番号③「側面ユニット支柱」についてです。
私の購入したキットでは、側面ユニット支柱と思われるアクリルパーツが４つ入っていたのですが、よく見ると、４つの内３つが全く同じ形状で、一つだけ異なる形状のアクリルが入っていました。
※したがって、[この手順](https://github.com/Taro-Hayashi/KillerWhale/blob/main/%E5%B7%A6%E6%89%8B%E7%94%A8/2_%E3%83%99%E3%83%BC%E3%82%B9%E3%83%A6%E3%83%8B%E3%83%83%E3%83%88.md#%E6%94%AF%E6%9F%B1%E3%81%AE%E7%B5%84%E3%81%BF%E7%AB%8B%E3%81%A6)から先に進めていません。

ガイドの写真では、側面ユニット支柱は２種類あるように見えるため、本来であれば２種類のアクリルが２つずつ入っているのが正解かと思います。

このままだと組みあがらないため、アクリル不足しているアクリルだけ購入させていただけないでしょうか？

Taro-Hayashi, 2023-08-12:
ご不便をおかけして申し訳ありません。すぐに不足分を発送いたしますのでお手数ですがメールかTwitterのDM、BOOTHのメッセージで送付先と側面ユニットの大小どちらが1つしか入っていなかったかを教えていただけないでしょうか。よろしくお願いいたします。

nerinanarine, 2023-08-12:
早々の返信、ありがとうございます。
たった今、Twitterでフォローしました。
お手数ですが、フォローバックお願いします。

相互フォローでないと、メッセージが送れないようです。。。

closed 2023-08-12 by Taro-Hayashi