Taro-Hayashi/KillerWhale #5: [🐜 Bug Report]  The Raspberry Pi contacts are too small.

kcw-gruntsoftware, 2023-10-20:


kcw-gruntsoftware, 2023-10-21:
cc: @Taro-Hayashi

Taro-Hayashi, 2023-10-21:
As I answered in my message to BOOTH, the hole is narrower than usual to conthrough, and I will sell you a replacement if you need one.
You have also inquired to YushaKobo, but it is not good to inquire about the same matter in many different directions.
If you don't pull it out forcibly, there should be no problem. I'll close this thread.

closed 2023-10-21 by Taro-Hayashi

kcw-gruntsoftware, 2023-10-21:
Hi @Taro-Hayashi ,
I bought two halves and the problem exists with both. I think the issue is the board needs a re-design so that the RPi or the header jumper can easily fall into the holes.  All of the other contact/holes fit perfectly and the components were a delight to add to the board.

As of now, I have had to desolder and destroy 2 RPis because the contacts dont connect to the holes and don't complete the circuit.   The picture in the provided in this issue was because the post of the header was stuck in the too-small hole. Even when gently pulling the jumper and RPi, the lead came up.

Since this is a design fault, I hope you can send the replacement boards directly at no charge  as you have done so in other cases.