Taro-Hayashi/KillerWhale #6: Used the female header pins in the wrong spot...

klevermonicker, 2023-11-09:
So I sockets the RP2040 onto the BASE unit, thinking that's what was I supposed to do, and now I can no longer connect any of the modular pieces since I used that on the RP2040's (Part 19 on the base guide picture). Where can I get replacement sockets like this?

The part in question is #19: [here](https://github.com/Taro-Hayashi/KillerWhale/raw/main/img/base/IMG_4724.jpg)
I've purchased several other options and none are big enough to handle the square pins of the headers.  Can someone point me to the correct part?

Taro-Hayashi, 2023-11-10:
Hi,  this one is compatible with the pin headers included. But I cannot say with certainty that it can be used because of its different shape.
https://ja.aliexpress.com/item/1005004945845785.html
Please note that the height is 3.5mm.

Alternatively, you might use a pin header that fits your pin sockets.

I would charge you for shipping, but I could send you a bundled package, so please let me know if you need it.

closed 2023-11-11 by Taro-Hayashi

klevermonicker, 2023-11-13:
Thank you, I ended up desoldering the original sets and using them and putting one of my sets of sockets in place of the original. Thank you for the help and offer!!