Taro-Hayashi/KillerWhale #8: Electrostatic Discharge Damaged PCB

amknopes, 2024-01-11:
Hey! I was using my duo killer whale keyboard, and experienced a static shock on the metal toggle switch. After that, many of the keys don't work anymore. Some keys are always pressed, some don't work at all, some send the wrong keycode, and some activate multiple other keys. Can you add electrostatic discharge protection to the PCB?

Taro-Hayashi, 2024-01-12:
Hello, I hope you are not injured. I'm leaving the circuit protection to the Raspberry Pi Pico, any ideas?

amknopes, 2024-01-12:
I'm not hurt at all, thanks for your concern <3 
The damage would've happened between the toggle and the Raspberry Pi Pico. I think something as simple as connecting the toggle's metal to a diode-protected ground would work. Electrostatic discharge just needs a ground path to discharge into so that all the current doesn't go into the rest of the PCB. All the other components should be alright since the toggle is the only one that gets touched with regular use

Taro-Hayashi, 2024-01-16:
The outer metal part and the foot were not conducting, so I disassembled it and found it insulated. I don't think grounding in this condition will help much.
[image]
There is a 5-pin part that has legs on the case, but considering that the bottom acrylic also needs to be modified, it may be difficult to change it now.
[image]
I apologize for not being able to meet your request. We will continue to consider the possibility.

If you need the base unit board to replace the Raspberry Pi Pico, please contact me by email or BOOTH message so I can send it to you at your expense.
Thank you.

closed 2024-01-18 by Taro-Hayashi

amknopes, 2024-01-22:
Wow that's fascinating. The shock must've traveled from the casing through the air then. 

I definitely want to purchase another board. Since one keypress is registering as multiple for some keys, I suspect the damage might be in the top unit. Is there electrical testing I can do to determine whether the damage is in the base unit or top unit? 
Where can I find your email?

Taro-Hayashi, 2024-01-23:
here is my email. rinarinapokegogo at gmail.com. Please let me know the country name and I will calculate the shipping cost and get back to you. Payment will be made by paypal.

Registering keypresses as multiple may be found by examining the diode or path with a tester. I'll send you both the top and base PCBs, though I'll ask for more details if you need help figuring out the cause. And Please keep the switch plate and acrylic parts.