Taro-Hayashi/KillerWhale #10: ソースコードダウンロード

Rookie-999, 2024-01-18:
お世話になります。プログラム初心者です。
商品購入し、ソースコード(キーマップ)をいじりたく、qmkファームウェアをダウンロードしたいのですが、どこからダウンロードすれば良いでしょうか?
パスワードを一つのボタンに設定して、そのボタンを押されればパスワードを入力完了としたく、ダウンロード方法をご教示頂けますと幸いです。

宜しくお願い致します。

Taro-Hayashi, 2024-01-18:
ソースコードはこちらに公開しています。
https://github.com/Taro-Hayashi/qmk_firmware/tree/tarohayashi/keyboards/tarohayashi/killerwhale/
QMKにつきましてはこちらに導入手順がありますのでご覧ください。
https://docs.qmk.fm/#/ja/newbs_getting_started

closed 2024-01-18 by Taro-Hayashi

Rookie-999, 2024-01-18:
お世話になっております。
試したのですが、この商品のソースコードが入っていませんでした。
[image]
git clone https://github.com/Taro-Hayashi/qmk_firmware.git と入力してもソースコードは無く、どのようにすれば解決出来るかわかりません。
お手数おかけし申し訳御座いませんが、ご助力頂けますと幸いです。

宜しくお願い致します。

Taro-Hayashi, 2024-01-18:
ソースコードが入っているのはtarohayashiブランチになっています。通常のcloneではmasterブランチになっておりますのでブランチを切り替えてみてください。よろしくお願いいたします。

Rookie-999, 2024-01-18:
何度も申し訳御座いません。
ブランチの切り替えがうまくいかないようで、ソースコードが入手出来ません。
プログラム初心者で理解出来ておらず申し訳御座いません。

どこで切り替えを行えば良いでしょうか?

宜しくお願い致します。

Taro-Hayashi, 2024-01-18:
cloneでダウンロードしたディレクトリに移動して「git checkout tarohayashi」で切り替わると思います。直接ZipファイルをダウンロードしていただくかGitHubデスクトップをお使いいただくこともできます。
申し訳ありませんがGitやGitHubの使い方、コンパイル方法につきましては解説しているサイトをご参照いただければと存じます。よろしくお願いいたします。

Rookie-999, 2024-01-18:
お世話になっております。
ご教示有難う御座います。切り替えが出来ました。
お手数おかけし申し訳御座いませんでした。とても助かりました。

宜しくお願い致します。

Rookie-999, 2024-01-18:
お世話になっております。
何度も申し訳ございません。
mysysでmakeしたのですが、hexファイルは作成されませんか？