Taro-Hayashi/KillerWhale #14: Led not working

fernando-vasconcellos, 2024-03-29:
I recently bought this product, flashed the necessary uf2 and even though the keys are working, the leds are not lighting up.
[image]

In there you can see that is not consistent, also the connection between the top part and the rest of the board is done by those cables to be able to test stuff, but when it was fully assembled the situation was the same.
I tested the electric connection on the leds and they are connected.

How could I fix this?
Also, I would appreciate it if you would upload the schematics, since that could give me more insight on the path to follow

Taro-Hayashi, 2024-03-29:
Hi, fernando. Thank you for purchasing. 
The LED that is glowing is connected to the LED in the upper left corner. It is connected as shown in the image.
If that path, 5V, and GND are the correct voltages and it does not glow, the LED may be damaged and may work if replaced.
I've uploaded a schematic for your reference.
https://github.com/Taro-Hayashi/KillerWhale/releases/download/0.23.0/scheme_killer_whale.zip
[image]

NandoVasconcellos, 2024-03-29:
Thank you very much! I will definitely try this once I'm back :)
And also, big thank you for the schematics :)

closed 2024-03-29 by Taro-Hayashi

NandoVasconcellos, 2024-04-01:
Hey, I wanted to follow up on this.
I've been testing with a multimeter and I didn't find a problem with any led. I've been also using the others side top piece and it only lights up the 23th.
At this point I'm a little bit desperate, Any idea? :(

NandoVasconcellos, 2024-04-02:
So, I've tried to change the firmware, reset completely the raspberry pi, I also changed the position of the jumpers just in case I made a mistake (that's why in the photo you see the jumper in "none" position).
Nothing of this worked so far.

Taro-Hayashi, 2024-04-03:
The LEDs are connected in series and the Raspberry Pi Pico decides how many LEDs to light up. Therefore, if the LEDs stop emitting light in the middle of a sequence, it means that the LEDs are not electrically connected or the number of LEDs is incorrectly specified.
If you have checked with a multimeter and there is no problem with the electrical connection, and you are using the firmware that I distribute, then the LEDs that are not glowing may be faulty. However, if the same phenomenon occurs on another top plate, it means that the LEDs in the same location have failed by chance, which is a bit unlikely.
This is not a solution, but if you need something in the form of a circuit board or parts, please contact me by DM on twitter(X) or email as I will ship it to you for about the manufactual cost and shipping.

fernando-vasconcellos, 2024-04-05:
Could you give me your email?
Or, if you prefer, this is mine fgvn18@gmail.com

I would be happy to request both top and base because neither of them are working, I've tried everything, even changing one of the raspberry pi pico with a brand new one and still, malfunctioning.

Taro-Hayashi, 2024-09-10:
Hi, I have received the amount and am working on it. I was going to contact you after it was shipped, but it is taking longer than expected. Sorry for the anxiety. 
Thank you.