Taro-Hayashi/KillerWhale #15: TRASSケーブルを接続してもUSBを繋いでいない方が

Okamoto0y, 2024-03-30:


Taro-Hayashi, 2024-03-30:
ちょっとそれだけだとわかりませんが導入されているファームウェアが左右接続用でないということはありませんか。トラックボールが認識されないのはベースユニットのジャンパーの指定が間違っている可能性があります。あとは基板や部品の故障でしょうか
交換用の基板の販売は可能ですので必要な場合はどのユニットが必要かをメールかBOOTHのメッセージにご連絡ください

Okamoto0y, 2024-03-31:
ご回答ありがとうございます。

■ファームウェアについて
ファームウェアは、ご準備いただいている
左右分割 右手ボール [tarohayashi_killerwhale_duo_ballright.uf2](https://github.com/Taro-Hayashi/KillerWhale/releases/latest/download/tarohayashi_killerwhale_duo_ballright.uf2)
こちらをDLして割り当てております。

割り当ての際は、
・TRSSケーブルを抜いた状態でBOOTSELボタン押下した状態でケーブルを接続
・DLしたファームウェアをドラッグ。これを左右どちらにも実施で行っております。

こちらの方法は、問題はなさそうでしょうか？

ちなみにUSBでどちらも接続したらPCには認識されますね。
どうしてもTRSSを繋ぐと認識しなくなります。ケーブル買ったばかりですが、初期不良ですかね？

■トラックボールについて
基盤上にある「TARACKBALL」の箇所にジャンパーの指定はしているのでこちらも問題ないかとは思います、、、、
トラックボールがジャンパーを触ると、認識はするようになりましたが
カクカクと画面の下にジャンプしてネット環境が重いときのマウスのようになります。
これはREMAPとかで修正できたりするんでしょうか？

Taro-Hayashi, 2024-03-31:
ご利用のファームウェアや更新方法は問題ないですね。テスト用のファームウェアも使ってみてください。
ケーブルの不良かどうかはテスターで両端の導通を調べたりオーディオに繋いで音楽を聞いてみる、他のキーボードで使ってみる等で検証できます。
ジャンパーを触ると認識するようになるということはジャンパーの接触に問題があるかもしれません。その症状ですと
Remapとは無関係で、ジャンパーの接触不良で飛び飛びになっているかセンサーが故障してる可能性があります。
メイン基板やラズパイが故障した可能性もあると思います。

Okamoto0y, 2024-03-31:
承知いたしました。では下記の方法試してみます。
・初期化を行い、テスト用のファームウェアを導入して動作確認も行ってみます。
・TRSSに関してはスピーカーなどで試してみます

・上記でダメな時、ジャンパーを購入して調査
　→ これでも解消できない場合は、基盤だと考えれるので、基盤購入します。

メイン基盤の購入の場合はどのように行えばいいでしょうか？

Taro-Hayashi, 2024-03-31:
BOOTHで販売いたしますので必要になりましたらご連絡ください。

Okamoto0y, 2024-03-31:
承知いたしました！TRSSケーブルの不備はないことを確認しました

テスト用のファームフェアを導入したのですが、それでもやっぱりトラックボールの反応が悪いですね、、、、
ジャンパー購入して動作確認します

Amazonで2mm、2.54mmどちらを購入したらいいとかありますか？

Taro-Hayashi, 2024-03-31:
2.54mmです。

Okamoto0y, 2024-03-31:
承知いたしました。ありがとうございます。
では一旦こちらクローズしていただき購入が必要になった際に改めてご連絡させていただきますのでよろしくお願いします。

closed 2024-03-31 by Taro-Hayashi