Taro-Hayashi/KillerWhale #16: オートマウス機能の移行、復帰の感度値の設定機能

tmyymmt, 2024-05-09:
とても良いキーボードのキット公開、ありがとうございます。この仕組みで立体にするのも凄い発想だと思いました。
左をジョイスティック側面ユニット、右をトラックボール側面ユニットで、2台を繋げて左右分割キーボードとして使用しています。
左のジョイスティックは普段は十字キー動作で、たまにマウス動作に切り替えて使っています。

さて、トラックボールを動かしたときに一時的にLayer7に切り替わるオートマウス機能について、最近のファームウェアだとF(J)キーを押してもクリックとして認識されないことが多くなり困っています。

確か、[0.23.0](https://github.com/Taro-Hayashi/KillerWhale/releases/tag/0.23.0)くらいまではトラックボールを動かした後に少し長い時間があった気がするのですが、[0.23.9.1](https://github.com/Taro-Hayashi/KillerWhale/releases/tag/0.23.9.1)では少しも時間がないようになった気がします。
また、[0.24.7](https://github.com/Taro-Hayashi/KillerWhale/releases/tag/0.24.7)では、トラックボールを動かした時にLayer7に切り替わるまでに必要な移動量（時間？）が大きくなった気もします。

オートマウス機能の、Layer7への移行と、Layer7からの復帰の、感度（移動量か時間の閾値？）を微調整できるようになりますでしょうか？

よろしくお願いいたします。m(_ _)m

Taro-Hayashi, 2024-05-10:
こんにちは、オートマウスにつきましては設定項目がいくつかあり、レイヤーを切り替える移動量、切り替える時間、レイヤーが戻った後にまた変更可能になるまでの時間、マウスに関係ないキーを使った時に切り替えが起こらないようにする時間がございます。

最適な値を模索中でバージョンによって概ねおっしゃる通りの変更内容になっています。更新に伴い切り替える時間を増やしてみましたが何かご提案があればお知らせください。

現状ではキーボードを使用中にはそれらの値を変更することができず、また利用方法や好みによる部分もあるかと存じますのでよろしければファームウェアの作成もお試しいただければ幸いです。

よろしくお願いいたします。

tmyymmt, 2024-05-11:
おぉ！早速のご対応ありがとうございます！！

F(J)キーがクリックとして認識されるようになりました！

そうなんですね。使用中に変更することができないとは。そしたら動的な調整は難しそうですね。

確かにこの部分は好みだと思いますので、ファームウェアを作成して値を色々いじってみようかと。設定値のコメントもありがとうございます。助かります！
https://github.com/Taro-Hayashi/qmk_firmware/blob/tarohayashi/keyboards/tarohayashi/killerwhale/duo/config.h

もし何か上手い策が見つかりましたらご連絡いたします。

ありがとうございます！

closed 2024-05-11 by tmyymmt