Taro-Hayashi/KillerWhale #17: 追加ユニットの左右の手順が逆だと思われる

Nnogu568, 2024-05-11:
本ビルドガイドを使ってkillerwhaleを作成しているものです。
おそらく昨日5/11あたりで手順が差し替えがされたと思っています。(細かいcommit履歴までは追っていないです。)
追加ユニットの手順
https://github.com/Taro-Hayashi/KillerWhale/blob/main/%E5%8F%B3%E6%89%8B%E7%94%A8/5_%E8%BF%BD%E5%8A%A0%E3%83%A6%E3%83%8B%E3%83%83%E3%83%88.md
例えば上記は最新の右手の追加ユニットの手順かと思いますが、以前の手順と見比べると「みぎのうら」と書かれている面と「ひだりのうら」と書かれている面ではんだ付けしている部分が異なるようにみえます。
まだ、killerwhaleを作成中なので勘違いだったらすいません。

Taro-Hayashi, 2024-05-11:
ご指摘ありがとうございます。先ほど修正いたしました。
写真が原因で組み立てに不都合が出てしまいましたら交換品を送付いたしますのでご連絡ください。

Nnogu568, 2024-05-12:
修正ありがとうございます。
組み立てには特に影響出ていませんが、作成中のためまた何かあればご連絡させてください

closed 2024-05-12 by Nnogu568

Nnogu568, 2024-05-18:
本件とは関係ないですがジャンパーが一つ足りず完成できませんでした。
何かご対応お願いできますでしょうか。

Taro-Hayashi, 2024-05-19:
大変失礼しました。お手数ですがメールかBOOTHのメッセージ等で送付先をお知らせいただいてもよろしいでしょうか。