Taro-Hayashi/KillerWhale #18: Possible error in diode direction on trackball unit guide

Jorres, 2024-05-16:
Hello! I apologize for texting in English, but my Japanese isn't good yet. 

I'm following the right section, trackball unit page: https://github.com/Taro-Hayashi/KillerWhale/blob/main/rightside/3_SIDE_TRACKBALL.md#soldering-of-the-diodes-back-side

I have bought a unit very recently, 1-2 weeks ago. On my part, most of the diodes have the same direction as on the part in the guide, except for the two diodes. Please have a look at these two pictures: the first picture from the guide, the second is the photo of the part that I have.  

Which one is correct? I have assembled according to the markup on the part, and the right part of the keyboard doesn't work yet. I want to rule out this diode misdirection as a possible issue. Can you please confirm which one is correct?

Here is the image from the guide, I have put an arrow where the black end of the diode should go:
[image]

Here is the part I bought:
[image]

Jorres, 2024-05-16:
Apologies for the pictures of slightly bad quality, for some reason my phone stopped focusing clearly enough :( I hope this will be enough for the debug.

Taro-Hayashi, 2024-05-16:
Hi, sorry for the confusion, the PCB print is correct. Please install as per the PCB print (the diode mark is in the opposite direction to C3).

This is happening because when I recently modified the PCB and build guide, I changed the orientation of the diodes because of the wiring.

The build guide for that version is here.
https://github.com/Taro-Hayashi/KillerWhale/tree/0.24.7
Trackball SIDE Unit (right side):
https://github.com/Taro-Hayashi/KillerWhale/blob/0.24.7/rightside/3_SIDE_TRACKBALL.md
Also, here is an image that shows the orientation of the relevant diode.
[image]

Thank you.

Jorres, 2024-05-17:
Thank you for the quick response. I was able to find my manual mistake elsewhere, not related to these diode positions. Closing the issue!

closed 2024-05-17 by Jorres