Taro-Hayashi/KillerWhale #20: Remap用ファームウェアでエラーが出る

gla66, 2024-05-26:
Remap用ファームウェアを書き込むとキーボードが認識されず、キー配置の変更や左右分割キーボードとして使用できません。
どのようにすれば解決できますでしょうか。

テスト用ファームウェアではKiller Whale Soloとしてデバイス認識されますが、
Remap用ファームウェア単体用ではPC上でドライブエラー表示がされ、
Remap用ファームウェア左右分割用ではそもそもデバイスとして認識されません。

Taro-Hayashi, 2024-05-26:
Raspberry Pi Picoの裏表はあっていますか？
また、テスト用ファームウェアはRemapに対応しておりませんので通常はデバイス選択には出てきません。ドライブエラーが表示されるとのことで正常にファームウェアがインストール出来ていない可能性がございます。お使いのPCのUSB周りに既知の不具合がないかご確認ください。ハブを用いずに直接PCに接続する、USBの差込口を変更する等していただくと違う結果が出ることがございます。
こちらの一番下、"Resetting Flash memory"にRaspberry Pi Picoのリセット方法が書いてありますのでこちらもお試しいただくと改善する可能性がございます。
https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html
交換用の基板がご入用の場合は販売いたしますので必要なユニットをメールかBOOTHのメッセージにご連絡ください。

gla66, 2024-05-26:
ハブを使って接続していたことが原因でした。直接接続したところ、全て問題なく機能致しました。
お手数をおかけいたしました。
迅速なご回答と素敵なキーボードありがとうございます。

closed 2024-05-26 by gla66