Taro-Hayashi/KillerWhale #21: Question about reflashing

Jorres, 2024-06-03:
Hi! Thank you for designing such an awesome keyboard. Second week in, and I'm enjoying it very much!

I would like to ask for an advice regarding flashing the firmware.

I'm writing the code for the firmware myself using QMK. I usually reflash the keyboard multiple times per day when changing something and trying something new. Because I flash very often, I would like to not break the usb port too soon (I know it is designed for ten thousand plug\unplugs, but I'm honestly a bit paranoid, the port looks very brittle).

When I recompile just the layout, I usually don't need to re-flash the slave side (there are very little cases when the slave side must also be reflashed). 

Usually, to put the Pi Zero into bootloader mode, you need to unplug the cable, hold BOOTSEL (or the special bootmagic key) and plug the cable back again. I have found a way to put the Pi Zero into bootloader mode WITHOUT unplugging and then replugging the cable (using `reset_usb_boot` function from pico library, can provide code if necessary). The only question remaining is the TRRS cable. 

I know I must not plug or unplug TRRS cable while the keyboard is connected via USB, this makes sense, I can short-curcuit the board. I'm not going to do that. But, can I put the keyboard into bootloader mode and flash it WITHOUT unplugging TRRS cable? In theory this must work, because it should be equivalent to just unplugging the usb cable and plugging it back in (which is safe). But I'm not willing to experiment, this is an expensive toy and I don't want to short circuit something accidentally. But since you designed the board, maybe you know more definite answer to this problem?

If what I'm thinking is correct, I can reflash the master-half of my keyboard as many times as I want, without unplugging either usb or TRRS cable at all. This means maximum lifespan for both usb and TRRS ports!

As a sidenote: maybe it will be useful to add some key combination that would drop the keyboard into bootloader mode? Usually people plug and unplug the cable in the first couple of days very much, surely it will be more convenient to give the option of not unplugging the cable when reflashing?

Sorry for a very long issue! :) Hope it's not too much trouble.

Taro-Hayashi, 2024-06-04:
Hi, thanks for using Killer Whale. I'm using a USB cable with a magnetic connection and did not notice the inconvenience.

Could you get into the bootloader with QK_BOOTLOADER, it is here in Remap.
[image]

I don't know exactly, but I always Flash with the left and right side connected with the TRRS cable and never had a problem.

I would be glad if I could help.

Jorres, 2024-06-04:
Thank you for the reply. Since you say you don't unplug TRRS cable, I also decided gave it a try, and thankfully, everything worked, I didn't need to unplug any of my cables. And since I was only reflashing the layout, I don't need to reflash the slave. This is cool, I can basically reflash my layout infinite times without ever touching the cables :)

Also confirmed that QK_BOOTLOADER works, I didn't need any custom Pico code after all. Not a big deal but -10 redundant lines of code is always nice :)

All my questions answered, closing the issue

closed 2024-06-04 by Jorres