Taro-Hayashi/KillerWhale #22: QMK MSYS Compile Error

amknopes, 2024-07-23:
I just grabbed the latest killer whale duo code from [here](https://github.com/Taro-Hayashi/qmk_firmware/tree/tarohayashi/keyboards/tarohayashi/killerwhale/duo) and am getting a compiler error for it when using QMK MSYS:

[image]

I'm not certain if this is an issue with the code or with my QMK installation

Taro-Hayashi, 2024-07-23:
Hi, I'm sorry.
The code I have published is for qmk firmware version 0.24.8, but while I was playing around with it, I bumped it up to 0.25, and I think that's why it won't compile.
If you change the qmk version to 0.24.8, it will probably pass. I'm trying to catch up with the latest, so please wait a bit.

amknopes, 2024-07-24:
okay I'll try downgrading my qmk, thanks!

closed 2024-07-24 by Taro-Hayashi