Taro-Hayashi/KillerWhale #23: FW Version 0.24.8 losses TRSS Comms when changing RGB Colors

shiori1425, 2024-07-24:
I built my duo killer December last year. I'm currently using two d-pads with add units on both sides.
Today I upgraded the firmware to version 0.24.8 (tarohayashi_killerwhale_duo_via.uf2) on both sides. 

Immediately after the firmware update things are working fine with the default config. 
I then go to remap and flash my personal keyboard layout and things continue to work fine. 

The problems start when I change the RGB colors, either before remap using the keyboard itself or through remap, whenever I change the leds the TRSS communications stops working. It starts getting very laggy typing across the two sides (works fine with just the usb side) and occasionally the non-usb side will just completely stops working. 

Redownloading the 0.24.8 firmware fixes the issue and I don't have a big problem with the default red color but I would like to be able to change it at some point. 

Any ideas why this might be happening? I don't know how to build the uf2 files to debug this myself sadly. Might just be time to learn.

Taro-Hayashi, 2024-07-24:
Not sure why if it works correctly with older firmware, but If the firmware on both hands has been updated it could be a power problem. Try connecting directly to a PC USB port without using a hub or changing the port or cable to be connected. It could also be caused by LEDs, MOSFETs, or resistors that are not soldered perfectly.
Lighting can be set not only with Remap but also with the keys on the keyboard (RGB keys in the DEVICE tab). Try that as well, and the problem may be fixed once.

shiori1425, 2024-07-25:
I don't think it's the soldering. It was working perfectly prior to updating and I do circuit board work professionally. The keyboard is not connected through a hub but direct to a usb 3.0 port on the mobo. The issue seems to not happen as much with the keyboard controls and I'll keep messing with that to see if I can get steps to reproduce the issue without using remap.app.

The issue is pretty consistently repeatable when using remap.app:
  1. Download 0.28.4
  2. Use Remap to flash my keymap
  3. Use Remap to change LED color
  4. Non-usb side stops working. 
  5. Redownload the 0.28.4 uf2 and everything is working again.

Taro-Hayashi, 2024-07-25:
I can't reproduce it on my end and don't know how else to deal with it. If anyone has the same symptoms or knows how to solve it, please let me know.

shiori1425, 2024-07-30:
I can't reproduce the issue using just the default keyboard led controls. I guess the issue is with remap.app and not the firmware. 

I'll close the issue here and bring this up on the remap discord.

closed 2024-07-30 by shiori1425