Taro-Hayashi/KillerWhale #24: Bad or misleading instructions

asdf23, 2024-07-29:
On the page: https://github.com/Taro-Hayashi/KillerWhale/blob/main/leftside/3_SIDE_DPAD.md

The directions say:
"Insert the diodes in the 4 locations on the back side of the PCB, and solder from the front side of the PCB, and cut the diode legs. "

The image shows 8 diodes
https://github.com/Taro-Hayashi/KillerWhale/raw/main/img/3_2_dpad_l/3_1_5_diodes.jpg

And the kit comes with 5 diodes ... ?

Do I need more diodes and the instructions/picture is incorrect?

Harshadb, 2024-07-30:
(Commenting as someone who built the KillerWhale.)
You are correct, since the instructions are partly from other side units and also translated.
The trackball & wheel side units use 4 diodes. The Joystick unit uses 5. The D-Pad unit uses 8.
(The japanese version of the D-Pad instructions has the correct number: 8)
Check the other side unit packs to confirm the number of diodes.

I am sure Hayashi-san will clarify and answer you.

Taro-Hayashi, 2024-07-30:
Thanks for telling me that instructions is wrong. 8 diodes is correct.
If a kit contains only 5 diodes, it is a packaging error. I apologize again.

Depending on where you live, it may be inefficient to send only the diodes, so may I ask you to purchase 1N4148 from elsewhere? I would pay up to $0.2 each($0.6)  plus shipping.

Thank you.

asdf23, 2024-08-04:
I am so sorry, I was looking at the wrong board. It took me a while to see it. I see now that I had opened the board for the track ball. 
I did order some extra diodes and in the process of connecting these I found the mistake.

closed 2024-08-04 by asdf23