Taro-Hayashi/KillerWhale #25: TRRS ports specification

Jorres, 2024-09-06:
Hi! I would like a piece of advice about TRRS ports.

I was building a custom TRRS cable and accidentally built one with 2.5mm jacks instead of 3.5mm. (all my fault, of course, haha :) 

Given that KillerWhale has two sets of TRRS ports, each side, I started thinking about whether I could replace one 3.5 port with 2.5 port on each side. So that I could use my newly built 2.5mm jacks :)  

This should be theoretically possible if I manage to find the TRRS ports with exact same dimensions and pins. I started looking and found many seemingly identical ones. Could I please ask you to specify the exact name of TRRS port that you put in KillerWhale sets? Right now, both Japanese and English documentation just says "TRRS jacks": 

[image]

Thank you so much for designing such a cool keyboard! Happily using it for almost half a year already and loving it very much!

Taro-Hayashi, 2024-09-07:
Hi, Jorres. Thank you for message.
PJ-320A or I think you could use something similar in shape.

Jorres, 2024-09-07:
Hi! Thank you very much, it seems to be it.

closed 2024-09-07 by Jorres