Taro-Hayashi/KillerWhale #26: RGB light vs RGB matrix?

Jorres, 2024-09-09:
Hello! 

I'm very sorry for posting my third question already and taking much of your time. 

In QMK, there are two distinct features: [RGB matrix](https://docs.qmk.fm/features/rgb_matrix) and [RGB light](https://docs.qmk.fm/features/rgblight). In KillerWhale, all RGB diodes (both under the keys, and a few ones under the keyboard) are configured as RGB light (basically, QMK treats all 66 diodes as one contiguous LED strip). 

I can color different parts of the strip in my own colors, as in this example:

```
const rgblight_segment_t PROGMEM base_layer[] = RGBLIGHT_LAYER_SEGMENTS(
        {0, 3, HSV_PURPLE},
        {3, 24, HSV_RED},
        {27, 1, HSV_PURPLE},
        {28, 2, HSV_RED},
);
```

produces the following lighting on left half:

[image]

But it has one very inconvenient consequence: I can control only color of the diodes programmatically, [such as HSV, for example](https://docs.qmk.fm/features/rgblight#functions), but I can not change brightness and on\off status. Basically, the whole keyboard either is lit, or the whole keyboard is un-lit. 

This makes a lot of interesting usecases impossible. I will try to name a few:

- light only 3 diodes in the bottom of the keyboard for a nice glow, while disabling all the remaining ones
- make a layer, where only h\j\k\l would glow, a nice vim-like feeling
- making complex animations that require turning on\off individual diodes or changing their brightness

Now, the bad part: I'm very bad at hardware, so I don't even understand if it's theoretically possible to change RGB light to RGB matrix programmatically. Maybe it even is impossible due to how wiring has been designed on Killerwhale. So my question is:

1. Maybe I'm wrong and there is a way to turn on\off diodes when using RGB light?
2. Why did you go for RGB light instead of RGB matrix? Do you think I can reprogram the keyboard to support RGB matrix, or will it require hardware modifications (and thus basically impossible)?

Taro-Hayashi, 2024-09-10:
Hi, 
Although RGB Matrix can be supported, it cannot be used in the firmware to be distributed for the following reasons
- RGB Matrix is not supported by REMAP.
- Firmware for top unit with wheel must be prepared separately because the LED numbers are misaligned due to the different number of keys.

To turn off the LEDs individually, I would have to set the value of V to 0 or specify HSV_BLACK.

Thank you in advance.

Jorres, 2024-09-10:
Thank you for your response. Indeed, I didn't understand how HSV worked. For simple usecases turning leds off with HSV_BLACK worked like a charm! 

Here is how lower diodes only glow in the dark, for example: 
[image]

I will test later whether this way is fast enough for complex animations, so that the keyboard still feels responsive. But even if not, a lot of simpler cases can be solved with this workaround, so I'm happy with it.

closed 2024-09-10 by Jorres