Taro-Hayashi/KillerWhale #27: raspberry pi pico compatibility?

soemarko, 2024-11-20:
So, I bought this kit from Yushakobo, followed the instruction, flash the board. And then do the mainboard, after soldering the LED, it's not turning on when plugged in. I'm guessing that I'm somehow fried the pi.

The question is, can I use any pi pico? From pictures in local marketplace, it looked different.

[image]

Taro-Hayashi, 2024-11-20:
Hi, this kit is not compatible with the Raspberry Pi Pico compatible in the image. Some compatibles have lost VREF pin to increase analog IO to 4 channels and are not fully compatible with Raspberry Pi Pico.
I can sell the relevant PCB, mounting components and type-C Raspberry Pi Pico for ¥2000 plus shipping. Please contact me at my email address (rinarinapokegogo at gmail.com) with the name of the country and city I would sending it to and I'll calculate the shipping costs and get back to you with an overall quote. If you need other unit, I can sell them together.
Thanks.

soemarko, 2024-11-21:
I see. So, none of the pi pico on aliexpress, etc would be compatible?

Let me try something first, I should've socketed the pi. It's going to be hard to desolder it.

Taro-Hayashi, 2024-11-21:
These are confirmed to work.
https://ja.aliexpress.com/item/1005007299070475.html
https://ja.aliexpress.com/item/1005004937441045.html

Hope all goes well.

soemarko, 2024-11-24:
Yeah something is off. Took me awhile to desolder the pi off the pcb, but it is okay withuout the pcb.

[image]

I need to investigate further. Definitely need to find the socket for the pi first.

closed 2024-11-24 by soemarko