Taro-Hayashi/KillerWhale #28: Additional unit scroll wheel not scrolling?

soemarko, 2024-12-14:
I've check continuity with my multimeter, everything is connected correctly, to pin 20, 21 and GND. I've tried with another encoder (since I got two kits) also no luck. What could be the cause of this?

Taro-Hayashi, 2024-12-14:
Sorry, there seems to have been a problem with the firmware. Could you try this one?
https://github.com/Taro-Hayashi/KillerWhale/releases/tag/0.27.1

soemarko, 2024-12-15:
hmm, this is odd. a lot have changed, but the wheel scroll a little bit when first plugged. and now the trackball becomes scroll?

Taro-Hayashi, 2024-12-15:
Maybe that's how it's set up to scroll with the trackball, you can use the Mode Change key to cycle the function of operating the trackball from cursor movement to scrolling or keystrokes.
[image]

Or you could try resetting the Raspberry Pi Pico.
https://datasheets.raspberrypi.com/soft/flash_nuke.uf2

soemarko, 2024-12-25:
I can't figure this out. I may have shorted something, just don't have the time to keep debugging. Going to power through by building the left side and see if I can salvage the keyboard. Merry Christmas.

closed 2024-12-25 by soemarko

Taro-Hayashi, 2024-12-25:
I hope it goes well. Merry Christmas.