Taro-Hayashi/KillerWhale #29: USER 13	QK_USER_13	押している間だけ十字キー入力モードについて

mako3313, 2025-01-10:
いつもお世話になっております。
タイトルですが
duo用
common_killerwhale.c内を
/* キースキャン */
// マウスキー登録
bool is_mouse_record_kb(uint16_t keycode, keyrecord_t* record) {
    switch(keycode) {
        case MOD_SCRL: // 一時的にスクロール
            return true;
        case QK_USER_4: // 一時的に減速
            return true;
        case QK_USER_9: // マウス1 / 長押しスロー
            return true;
        case QK_USER_10: // マウス2 / 長押しスロー
            return true;
        case QK_USER_11: // マウス3 / 長押しスロー
            return true;
        case QK_USER_12: // 一時的にカーソル移動
            return true;
        case QK_USER_13: // 押している間だけ十字キー入力モード
            return true;
        default:
            return false;
    }

    return  is_mouse_record_user(keycode, record);
}　
現状上記の様に変更して使用させていたただいています。
 case QK_USER_13: // 押している間だけ十字キー入力モード
            return true;
を追加したことによって問題になることはありますでしょうか？
ご確認のほどよろしくお願いします。

Taro-Hayashi, 2025-01-10:
お世話になっております。
QK_USER_13を押すことでオートマウスが反応するようになりますがそれを問題と捉えるかどうかかと存じます。
他に影響が出ることは今のところ思い当たりません。
よろしくお願いいたします。

mako3313, 2025-01-10:
ご確認ありがとうございました。
今後もよろしくお願いします。

closed 2025-01-10 by mako3313