Taro-Hayashi/KillerWhale #31: Some LEDs do not light up on top part

mariusursache, 2025-03-25:
@Taro-Hayashi, as you can see, only few of the LEDs light up. From a previous closed issue on this topic, I assume that an LED is broken and they are in series. However, after desoldering and replacing half of the LEDs surrounding the area, the issue still persists. The keys work flawlessly.

1. Can you share the order of the LEDs, on the board so that I can replace the exact dead LED?
2. What are the voltages that should be present on each LED and the pins connecting the top with the base?

Thanks!

[image]

Taro-Hayashi, 2025-03-25:
Hi, this is the order of the LEDs, it looks like there is a problem with 7 or 8.

[image]

The voltage of the LEDs is 5V. The marked leg of the LED is 0V at GND and the diagonal leg is 5V.
Regards,

mariusursache, 2025-03-25:
Thank you, that was fast! ;-)

mariusursache, 2025-03-26:
Works great now! First half is built!

[image]

closed 2025-03-26 by mariusursache

Taro-Hayashi, 2025-03-26:
It's so cool. I've actually never seen that case and printed ball case combination before.

mariusursache, 2025-03-26:
Yes it is, I love it! Now I am trying to find some good keycaps (not a fan of this profile, will try some XDA/DSA profiles because it's easier to mix the keycaps, while for Cherry/SDA I am restricted by the varying profile).