Taro-Hayashi/KillerWhale #33: Left OLED not working

mariusursache, 2025-04-01:
Finalized both halves and everything works great (waiting for my resin-printed DES keycaps).

The left-side OLED screen does not show everything (the right side does). Is this normal? The OLED screen works, so it's not an issue with it.

Taro-Hayashi, 2025-04-01:
What is the status of not showing everything?
The side with USB connected can switch between layer and stats display, but the side without USB connected can only display layers.
Thanks,

mariusursache, 2025-04-01:
Left-side screen is blank (off), does not show anything. It is part of the trackball assembly.

1. I connected only left side - problem persists
2. I flashed with nuke firmware, then again with duo firmware - problem persists
3. Soldering is fine

mariusursache, 2025-04-03:
Built a separate trackball and it works.

closed 2025-04-03 by mariusursache